roboticslab-uc3m/teo-follow-me #11: [FollowMe] Improving voice

rsantos88, 2016-06-29:
Propongo que el robot utilice en la aplicación un paquete de voz más claro (tanto en inglés como en español). Como esto puede ser subjetivo, propongo también (si fuese posible) hacer una votación de qué voz le parece mejor a todo el mundo. ¿estáis de acuerdo?

David-Estevez, 2016-06-30:
Posibles alternativas: **espeak**

Voz en inglés:
```
yarp read /tts:i | espeak -v en
```

Voz en castellano:

```
yarp read /tts:i | espeak -v es
```

Se podría incluso poner ambas en paralelo (un puerto para diálogos en castellano y otro en inglés):

```
yarp read /tts-es:i | espeak -v es
yarp read /tts-en:i | espeak -v en
```

rsantos88, 2016-07-18:
Gracias @David-Estevez. He estado probando el sintetizador que viene con eSpeak. 
Aquí dejo un [ejemplo](http://espeak.sourceforge.net/samples/raven.ogg) de como pronunciaría.
Personalmente, me gusta mucho más la pronunciación de [Mbrola Voices](http://espeak.sourceforge.net/mbrola.html), las cuales se pueden utilizar con eSpeak.
Aquí os dejo ejemplos tanto en inglés como en español:
* [Inglés](http://tcts.fpms.ac.be/synthesis/mbrola/demo/en1.wav)
* [Español](http://tcts.fpms.ac.be/synthesis/mbrola/demo/es1.wav)

¿Qué tal os parecen? :smile:

smcdiaz, 2016-07-18:
Prefiero la pronunciación del Mbrola

rsantos88, 2016-07-19:
Aquí os dejo el resultado de las voces que os he comentado más arriba aplicadas a las aplicaciones actuales, para que sea más fácil asociarlo:
* [Follow me](https://drive.google.com/a/uc3m.es/file/d/0BxR76I90oKSmeDMxYkRpZzJGdG8/view?usp=sharing) (130 speed in words-per-minute)
* [Camarero](https://drive.google.com/a/uc3m.es/file/d/0BxR76I90oKSmamxFSFRqUDdVbHc/view?usp=sharing) (150 speed in words-per-minute)

¿Os parece bien entonces integrar estas?

RaulFdzbis, 2016-07-19:
A mi personalmente me gusta más la primer opción de Follow me. De todas formas podría estar bien poner ambas en castellano (e ingles si es posible también) para comparar correctamente.

rsantos88, 2016-07-19:
Gracias @RaulFdzbis! Cada idioma dispone de uno o varios paquetes de síntesis de voz distintos por lo que no se pueden poner ambas voces en castellano y en inglés.
Para Mbrola Voices dispones únicamente de los siguientes paquetes de voz en español neutro:
* [Español 1](http://tcts.fpms.ac.be/synthesis/mbrola/demo/es1.wav)
* [Español 2](http://tcts.fpms.ac.be/synthesis/mbrola/demo/es2.wav)
* [Español 3](http://tcts.fpms.ac.be/synthesis/mbrola/demo/es3.wav)
* [Español 4](http://tcts.fpms.ac.be/synthesis/mbrola/demo/es4.wav)

jgvictores, 2016-09-08:
Se ha tocado https://github.com/roboticslab-uc3m/teo-main/commit/51927f59f164420dceac83cd8a8d647d129c53d0, resultando en el punto 2 y 3 de la issue que ahora mismo es #13.

jgvictores, 2016-10-13:
https://github.com/roboticslab-uc3m/teo-main/issues/55

jgvictores, 2016-11-11:
Closed because https://github.com/roboticslab-uc3m/teo-head/issues/38 and roboticslab-uc3m/teo-main#55 are closed.

closed 2016-11-11 by jgvictores