roboticslab-uc3m/teo-follow-me #13: [ERRORS] Problems found in Follow-me demo

rsantos88, 2016-09-08:
Siguiendo el procedimiento de dicha Demo y después de varios intentos fallidos de hacerla funcionar correctamente, anoto los diferentes errores que se han ido encontrando:
* En diversas ocasiones se ha mostrado un error al cargar el módulo "followMeExecutionCore": 
`[ERR] (follow-me_App) cannot run followMeExecutionCore on localhost `
* El nuevo paquete del habla, ubicado en "teoBase_App",arranca correctamente la primera vez. Si se decide parar dicho módulo, al intentar arrancarlo de nuevo, mostrará el error:  
`[ERR] (teoBase_App) cannot run /sbin/runuser on /head : cannot run /sbin/runuser on /head`
La única solución para volver a arrancarlo es reiniciando el PC de la cabeza.
* Si se consigue arrancar con éxito el módulo del habla y se prueba en teoTools_App (escribiendo por teclado), funciona correctamente. Si arrancamos todos los módulos para hacer funcionar la Demo y establecemos las conexiones oportunas, no funcionará el habla que corresponde a la Demo (Teo no pronunciará las frases correspondientes al Follow me).
* Problemas con el movimiento de la cabeza: por 2 veces ha sido incapaz de funcionar correctamente. En la primera ocasión, la cabeza permanece rígida y no hay movimiento. En una segunda ocasión, solo aplicaba el movimiento de la cabeza en el eje vertical. No así en el horizontal.

jgvictores, 2016-09-08:
¿Pudrías cerrar esta issue y crear 3 issues específicos?

1. Necesitamos más datos para saber qué está ocurriendo con `followMeExecutionCore`: Attach to stdout y copiar output, lanzar desde consola y ver error...
2. y 3. Se pueden juntar en una issue que corresponde a [teo-main](https://github.com/roboticslab-uc3m/teo-main). Te adelanto que se debe al pipe (símbolo `|` en líneas 48 y 54 de https://github.com/roboticslab-uc3m/teo-main/commit/51927f59f164420dceac83cd8a8d647d129c53d0) para el nuevo sistema de habla consecuencia de #11.
4. Esto es cosa de [teo-body](https://github.com/roboticslab-uc3m/teo-body), probablemente hardware.

rsantos88, 2016-09-08:
La última efectivamente era un problema de hardware,  por lo que si el problema persiste abriré otra issue. Cierro por tanto ésta, en sustitución de las issues: https://github.com/roboticslab-uc3m/follow-me/issues/14 y https://github.com/roboticslab-uc3m/teo-main/issues/55 abiertas.
Saludos!

closed 2016-09-08 by rsantos88