roboticslab-uc3m/teo-follow-me #18: New follow-me with Carlos dialogue

rsantos88, 2016-11-15:
Adjunto el [fichero](https://github.com/roboticslab-uc3m/follow-me/files/592454/Dialogo.Follow.Me.v1.docx) con el diálogo que propuso Carlos en la reunión para las Demos.
La demo de follow-me actualmente se encuentra adaptada en inglés, con algunas diferencias (00cdb30ce2865b88e7f07b0a012564a043884cca) 
Falta incluir la parte en la cual TEO indica "estás a la derecha", estás a la izquierda", "estás en el centro".
¿Sería correcto llamar a la función [ttsSay](https://github.com/roboticslab-uc3m/follow-me/blob/develop/programs/followMeDialogueManager/StateMachine.cpp#L94-L108)  desde [InCvPort::onRead(Bottle& b)](https://github.com/roboticslab-uc3m/follow-me/blob/develop/programs/followMeExecutionCore/InCvPort.cpp#L18-L22)?

jgvictores, 2016-11-16:
Pese a ser más complicado, sería más conceptualmente correcto tenerlo dentro de la máquina de estados, y mejor de cara a un futuro con una máquina de estados más sofisticado.

rsantos88, 2016-11-30:
En ese caso, se llamaría a ttSay desde una hipotética función llamada saySites en la máquina de estados. Para ello, tendríamos que recibir la información de haarDetection a través de un puerto que conectaríamos con followMeDialogManager.
[image]
¿ Cómo lo ves @jgvictores ?

jgvictores, 2016-11-30:
Una conexión así tendría sentido (sin nombre)!

jgvictores, 2016-12-01:
Estuve pensando esto... ¿No sería más bien una conexión con launchManipulation la necesaria?
Para saber hacia dónde está apuntando el pan del cuello.

jgvictores, 2016-12-01:
Esto es, dialogue manager envia comando peroodicamente (si el estado lo permite),  y execution core comanda el movimiento en función del encoder de la cabeza.

rsantos88, 2016-12-05:
Entiendo que dialogue manager enviará un comando a través del puerto "outCmdPort" de la siguiente forma:
```C++
yarp::os::Bottle cmd;
cmd.addVocab(VOCAB_GET_POSITION);
outCmdPort->write(cmd);
```
Posteriormente, followMeExecutionCore deberá tener un OutGpPort.cpp y un OutGpPort.hpp (output get position port) el cual enviará  por un puerto yarp el valor del encoder del cuello.

Conformame @jgvictores  si esto sería correcto.

jgvictores, 2016-12-05:
No, dialogue manager podría enviar un:
```C++
yarp::os::Bottle cmd;
cmd.addVocab(VOCAB_WAVE_APPROPRIATE_HAND);
outCmdPort->write(cmd);
```
Y en realidad la clase [FollowMeExecutionCore](https://github.com/roboticslab-uc3m/follow-me/blob/develop/programs/followMeExecutionCore/FollowMeExecutionCore.hpp) hay que ampliarla para que además de `yarp::dev::IPositionControl *iPositionControl;` un `yarp::dev::IEncoders *iEncoders;`

rsantos88, 2016-12-05:
Ok, el código resultante quedaría de [esta forma](https://github.com/rsantos88/follow-me/blob/develop/programs/followMeExecutionCore/InSrPort.cpp). Faltaría programar la parte que permita el movimiento de cada brazo. Me gustaría saber como podría continuar con la parte del habla y cómo obtener los valores de los encoders de la cabeza desde el state machine para llamar a la función ttSay.

Un saludo y gracias @jgvictores

jgvictores, 2016-12-06:
Ok, hice fork de tu fork. [Aquí](https://github.com/jgvictores/follow-me/commit/3e1ce996a7cd36cc09683cf09b75e06428121c07) corregí algo que me parecía peligroso en cuanto al uso de punteros sin inicializar, y [aquí](https://github.com/jgvictores/follow-me/commit/f63ffe4be3dc02d924ddbde10ec68b50ecae2489) implementé `asrListenWithPeriodicWave()`. Si puedes probar el branch `non-blocking-read` de [mi fork](https://github.com/jgvictores/follow-me) te lo agradezco. Si funciona, de ahí hago PR a tu fork, y de ahí hacemos PR de tu fork a la original (roboticslab-uc3m).

rsantos88, 2016-12-12:
Funciona todo correctamente  :thumbsup:

rsantos88, 2016-12-12:
Realizados los respectivos pull request (#23)
Acepta si lo ves correcto @jgvictores

jgvictores, 2016-12-12:
PR #23 aceptado.

rsantos88, 2016-12-12:
Añadiendo el followMeArmSwing... ¿quedaría correcto dejarlo así? 
[image]

jgvictores, 2016-12-12:
Ok, so, in sight of the latest [follow-me-app.png](https://github.com/roboticslab-uc3m/follow-me/blob/204142e6027cf1cff2f1fbf076f68dc347c8c7f6/doc/fig/follow-me-app.png)... there's some work to do on the execution side. We must choose between:

1. Merging followMeArmSwing into followMeExecutionCore
2. Renaming followMeArmSwing->followMeArmExecution and followMeExecutionCore->followMeHeadExecution

I'd go for (2) -> less elegant, more ports, but probably easier and more maintainable.

rsantos88, 2016-12-13:
Ok. Then, we'll go for (2) method.
It would be like this:
[image]
I think, we've to extend followMeArmExecution to include new movements with two arms.

jgvictores, 2016-12-13:
Ok, seems good enough. I'd recommend:
- [ ] Drawing followMeDialogueManager<->followMeArmExecution and followMeArmExecution<->launchManipulation bi-direcional.
- [ ] Split followMeHeadExecution command port in two, make them type RpcClient.
- [ ] Make followMeArmExecution and followMeHeadExecution command ports of type RpcServer.
- [ ] Update the port names (.../rpc:s for server, .../rpc:c for client) and app xml connections.

rsantos88, 2016-12-13:
Ok, I renamed followMeArmSwing->followMeArmExecution and followMeExecutionCore->followMeHeadExecution in the code ( #24) and I updated the [image](https://github.com/roboticslab-uc3m/follow-me/blob/develop/doc/fig/follow-me-app.png).
Do you see it right @jgvictores ?

jgvictores, 2016-12-13:
Looking very very good. Thanks a lot for all this work.

I've updated my comment above regarding moving from yarp::os::Port to yarp::os::RpcClient.

rsantos88, 2016-12-13:
Second point done! Please @jgvictores , make sure that all looks good (#25)  :smile:

rsantos88, 2016-12-14:
In point (3), I've changed command port to RpcServer port (826ae571d03187f16a8141e7a71adc69eff77a54) & (1d38ce6f175d564b05dac8da144c159a408e49c4) in followMeHeadExecution, but in followMeHeadExecution, I don't know what ports do you want to change @jgvictores

jgvictores, 2016-12-14:
I'm guessing you mean the arm execution, where we set up callbacks. 
So the code of "getting a callback" from: http://www.yarp.it/port_expert.html
Has to start looking more like "getting replies" section of the same tutorial (when it starts speaking about callbacks). 

Those are just some pointers; check out the tutorial, try to see how it corresponds to the code, and I'll get back to you.

rsantos88, 2016-12-14:
Vale, si no me equivoco lo que pretendes sería comunicar por un puerto RPC followMeDialogueManager con followMeHeadExecution de forma bidireccional. Esta comunicación se podría emplear para que followMeDialogueManager haga una petición para conocer la posición del encoder de la cabeza y followMeHeadExecution responda de forma instantánea utilizando el método descrito en "Getting replies".
¿Es correcto?

jgvictores, 2016-12-14:
Correcto!

rsantos88, 2016-12-16:
Se ha añadido un nuevo VOCAB, implementando el código de la siguiente [forma](https://github.com/rsantos88/follow-me/blob/develop/programs/followMeHeadExecution/InDialoguePortProcessor.cpp#L37-L48)
¿Estaría correcto así @jgvictores ?

jgvictores, 2016-12-16:
Bien, creo que se momento vale!

jgvictores, 2016-12-16:
Oh, I just remembered. I think it's best not to return `false` on the InDialoguePortProcessor::read(ConnectionReader& connection) method.
Best is to add a VOCAB_FAILED, send it, and return `true`.

rsantos88, 2016-12-16:
[Updated](https://github.com/rsantos88/follow-me/blob/develop/programs/followMeHeadExecution/InDialoguePortProcessor.cpp) with that modification

rsantos88, 2016-12-16:
Update the port names and app xml connections.(93ec46456184381c99572c8dac718f8d1350ddad)
Something wrong @jgvictores ??

jgvictores, 2016-12-16:
Everything looks very good. To strictly follow [the current connection diagram](https://github.com/roboticslab-uc3m/follow-me/blob/5db319388f03813b5eb7e062290b8a0bfed96cf5/doc/fig/follow-me-app.png), [this](https://github.com/rsantos88/follow-me/blob/93ec46456184381c99572c8dac718f8d1350ddad/share/app/follow-me/scripts/follow-me.xml#L10) and [this](https://github.com/rsantos88/follow-me/blob/93ec46456184381c99572c8dac718f8d1350ddad/share/app/follow-me/scripts/follow-me.xml#L15) would have to run on `/manipulation`, but this is actually a more practical setup, so it's good enough for me!

rsantos88, 2017-01-10:
Por el momento ya está implementado y funcionando. Cada 10 segundos, indica a través de una frase si estás a la derecha, a la izquierda o en el centro según si el encoder se posiciona en **enc>10**, **enc<-10** o **-10<enc<10** respectivamente ([link](https://github.com/rsantos88/follow-me/blob/develop/programs/followMeDialogueManager/StateMachine.cpp#L135-L146))
El siguiente paso sería implementar los movimientos en los brazos y ampliar más el dialogo si se quisiera con frases alternativas.

rsantos88, 2017-01-11:
Tal y como se encuentra programado ahora mismo, el robot indicaría si te encuentras a la derecha, a la izquierda o en el centro, dependiendo de la posición del encoder del cuello. Cada 10 segundos estaría repitiendo tu posición, desde el momento en que le ordenes que te siga y le digas tu nombre. 
El problema es que si el robot deja de detectar una cara, seguiría repitiendo "ahora estas a la derecha"... a la izquierda... etc cada 10 segundos, dependiendo de la última posición en la que haya quedado su cuello.
Sería ideal que robot pudiera dejar de pronunciar estas frases en el momento en que haarDetection dejase de detectar una cara o volver a pronunciarlas en caso de que vuelva a detectar una cara.
Lo dejo como sugerencia para una posible mejora futura ;)

rsantos88, 2017-01-12:
Añadido pull request #26

rsantos88, 2017-01-18:
Actualizado el gráfico a los nuevos cambios: commit (40dba08d752720700104c443b2e2a6e90c4559ff)
Con todo esto, se pueden dar por concluidas las mejoras del follow-me propuestas.
Cierro issue :)

closed 2017-01-18 by rsantos88

jgvictores, 2017-01-18:
@rsantos88 Thank you so much!! Muchas gracias por todo el curro y esfuerzo. 🥇 😄 🥇

rsantos88, 2017-01-18:
:blush: