roboticslab-uc3m/teo-follow-me #33: Must update follow-me-app.png

jgvictores, 2017-04-22:
Must update [doc/fig/follow-me-app.png](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c3b61c999836419587aef5d1d5586dd2bd/doc/fig/follow-me-app.png), from source [doc/fig/editable/follow-me-app.graphml](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c3b61c999836419587aef5d1d5586dd2bd/doc/fig/editable/follow-me-app.graphml), with new repo names: `yarp-devices`, `vision`, etc.

rsantos88, 2017-04-24:
Done it at (f2f116e57d5536974e46e81988dbb5fe330bea4b)
I'll close this issue!!

jgvictores, 2017-04-24:
Maybe green could be used for `speech` and we can put `speechRecognition.py` in green?

jgvictores, 2017-04-24:
PS: As well as `yarpdev --device Espeak`.

jgvictores, 2017-04-24:
Something like `speech @ /head`.

rsantos88, 2017-04-24:
updated it! (01e088935e9d6742b1028306dc89109a58119b7f)

jgvictores, 2017-04-24:
@rsantos88 Thanks, really great!

closed 2017-04-24 by jgvictores