roboticslab-uc3m/teo-follow-me #34: Proposal of renaming to teo-follow-me

jgvictores, 2017-04-22:
Proposal of renaming to teo-follow-me, as this repo has become very teo-specific.

...or make it less teo-specific!?

PeterBowman, 2017-09-10:
> ...or make it less teo-specific!?

I'm working on a 2D haarTraining app on roboticslab-uc3m/vision and its controller counterpart at kinematics-dynamics, with a much simpler purpose: fetch object centroids from the grabber device and make the robot move towards the target until it's deemed close enough by a proximity sensor (or some other signal), then close the gripper tool (if available). I feel it could be part of something bigger, and this repo already has ASR support.

PeterBowman, 2021-11-14:
> Proposal of renaming to teo-follow-me, as this repo has become very teo-specific.

I'm giving my +1 as it is specifically designed for TEO. Nothing stops us from reusing it on other *compatible* robots, though, in which case I think a teo- prefix would not bother me anyway.

jgvictores, 2021-11-17:
Okay, did the quick rename...

Apart from updating the README (atow, pending), what other changes should this involve (CMake variables, file renaming...)?

PeterBowman, 2021-11-17:
@jgvictores please leave it as is, I'm working on this repo right now. I'll take care as soon as it's ready.

PeterBowman, 2021-11-19:
Done at https://github.com/roboticslab-uc3m/teo-follow-me/commit/c911b29a4455bec6a0ca3801d8f56cce4b6c09db except [teo-follow-me-app.graphml](https://github.com/roboticslab-uc3m/teo-follow-me/blob/a8062e56fb56bfeb52fe96c91cf686dd6ae75657/doc/fig/editable/teo-follow-me-app.graphml) (note I did update [teo-follow-me-app.svg](https://github.com/roboticslab-uc3m/teo-follow-me/blob/a8062e56fb56bfeb52fe96c91cf686dd6ae75657/doc/fig/teo-follow-me-app.svg)).

closed 2021-11-19 by PeterBowman