roboticslab-uc3m/teo-follow-me #37: Error complilation

jmgarciah, 2018-04-04:
branch develop

the code should be update.

there are the typical problem with `class yarp::dev::IPositionControl` that now is `class yarp::dev::IPositionControl2`

jgvictores, 2018-04-04:
`IPositionControl` still exists. Could you please copy the exact compilation error?

jmgarciah, 2018-04-04:
[  8%] Building CXX object programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/FollowMeArmExecution.cpp.o
/home/teo/repos/follow-me/programs/followMeArmExecution/FollowMeArmExecution.cpp: In member function ‘virtual bool teo::FollowMeArmExecution::configure(yarp::os::ResourceFinder&)’:
/home/teo/repos/follow-me/programs/followMeArmExecution/FollowMeArmExecution.cpp:43:30: error: ‘class yarp::dev::IPositionControl’ has no member named ‘setPositionMode’
     leftArmIPositionControl->setPositionMode();
                              ^
/home/teo/repos/follow-me/programs/followMeArmExecution/FollowMeArmExecution.cpp:61:31: error: ‘class yarp::dev::IPositionControl’ has no member named ‘setPositionMode’
     rightArmIPositionControl->setPositionMode();
                               ^
programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/build.make:62: fallo en las instrucciones para el objetivo 'programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/FollowMeArmExecution.cpp.o'
make[2]: *** [programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/FollowMeArmExecution.cpp.o] Error 1
CMakeFiles/Makefile2:177: fallo en las instrucciones para el objetivo 'programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/all'
make[1]: *** [programs/followMeArmExecution/CMakeFiles/followMeArmExecution.dir/all] Error 2
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2

rsantos88, 2018-04-12:
Done at https://github.com/roboticslab-uc3m/follow-me/pull/39

jgvictores, 2018-04-14:
As commented above, `IPositionControl` to `IPositionControl2` was not the issue.

#36 erased the conflicting line (https://github.com/roboticslab-uc3m/follow-me/pull/39/files#diff-7e5b6080f5c174044e4adaa6acf7d75dL43) rather than replacing it by its modern equivalent.

Here's some code for inspiration: https://github.com/roboticslab-uc3m/asibot-hmi/blob/d0d18336d5400e9a906c0b9e0fbea5ba9131c5e3/programs/webInterface/WebResponder.cpp#L654-L657

rsantos88, 2018-04-16:
> As commented above, IPositionControl to IPositionControl2 was not the issue.

The problem is `setPositionMode()` is deprecated and not used. We use `IControlMode2` to set the mode and I take advantage of this to update the interfaces and substitute `IPositionControl` by `IPositionControl2`.

> #36 erased the conflicting line (https://github.com/roboticslab-uc3m/follow-me/pull/39/files#diff-7e5b6080f5c174044e4adaa6acf7d75dL43) rather than replacing it by its modern equivalent.

I don't understand what does it have to do #36 (maybe it's only an error). I think it's better to use `IPositionControl2`. It inherits all of `IPositionControl` and it's updated with functions using more parameters. It's the same changes that I do with `teo-self-presentation`.

> Here's some code for inspiration: https://github.com/roboticslab-uc3m/asibot-hmi/blob/d0d18336d5400e9a906c0b9e0fbea5ba9131c5e3/programs/webInterface/WebResponder.cpp#L654-L657

Updated with more modular code: https://github.com/roboticslab-uc3m/follow-me/pull/41 (PR)

jgvictores, 2018-04-16:
Cool, `IControlMode2->setControlModes` is the way to go. Merged #41, thanks!