roboticslab-uc3m/teo-follow-me #40: Add mute/unmute microphone to this demo

rsantos88, 2018-04-12:
The idea is mute the microphone when Teo is talking something and we don't want to capture the noise.

rsantos88, 2018-04-17:
I copy [this](https://github.com/roboticslab-uc3m/speech/issues/13#issuecomment-381989664) issue which involves this demo:
>Here we've a limitation. We can mute or unmute the microphone by software automatically, sending the demonstration commands to speechRecognition when the robot is speaking or in a moments when we know for sure we're not going to talk. But, the problem here is what happen when the robot is waiting for an order and at the same time is doing something.
E.g: Teo is following you but it's checking all the time if you are going to say "stop". Here, we can take the option to mute all the time the microphone, but we can't stop him by command voice. If we don't mute the microphone, Teo will receive to much noise and false orders.
>
>At this point, any possible idea of how to resolve that with the work done ?

rsantos88, 2018-04-17:
Possible solutions:
* As @PeterBowman propoused: 
>One thing I can come up with are acknowledgements. That is, make TEO send an Are you sure? notice back to you, and await confirmation. Is it feasible in your demo workflow?
* filter the noise with greater concatenation of words: E.g `please stop TEO` instead of `stop` but I want to get a solution to avoid modifying the dictionaries...

rsantos88, 2018-04-17:
Other idea (I like it):
1. (every 2 or 3 minutes) Teo: "If you want me to stop following you, please say: **stop** "
2. `setMic unmute` (wait 2 seconds)
-- You: "wait" -> `setMic mute` & Teo: "ok, I will stop following you..."
-- You: "...."   -> Teo continues following

rsantos88, 2018-04-18:
As I talked with @jgvictores , we can left this demostration with this changes ([see PR](https://github.com/roboticslab-uc3m/follow-me/pull/42)) by the moment: only mute the microphone when Teo is speaking but take into account that the problem [aforementioned](https://github.com/roboticslab-uc3m/follow-me/issues/40#issuecomment-382017078) still persists

Can I close this issue?

jgvictores, 2018-04-19:
Merged #42. Yes, feel free to close it.

I'll probably open new, specific issues.

closed 2018-05-07 by rsantos88