roboticslab-uc3m/teo-follow-me #43: followMeDialogueManager could benefit from some refactoring

jgvictores, 2018-04-19:
Certain parts and/or port/bottle naming can be improved, for instance: https://github.com/roboticslab-uc3m/follow-me/blob/b462dad42fc243732744ccb276249a341ce9c2d3/programs/followMeDialogueManager/FollowMeDialogueManager.cpp#L56-L96

As general practice:
- Port/bottle names should be clear.
- Bottles should be cleaned *before* use, not after.
- I'll add a new issue related to vocabs vs strings.

rsantos88, 2018-04-19:
Sorry for doing this modification directly in `develop` branch
Only I've modified the bottle names and the order of clear() function
Commit: https://github.com/roboticslab-uc3m/follow-me/commit/35aae0e507b1d493222b22c9abd6fcec0120d55f

jgvictores, 2018-04-20:
35aae0e507b1d493222b22c9abd6fcec0120d55f is good, thanks!

Closing issue.

closed 2018-04-20 by jgvictores