roboticslab-uc3m/teo-follow-me #48: Recognition of objects (besides faces)

rsantos88, 2019-03-28:
I propose the possibility (I propose myself to do it in the future) to do some demo based on recognition and tracking of objects (ex with TensorFlow). Surely someone has already thought about it, but it would be a good way to update and make this demo more interesting and interactive with voice commands ;)

jgvictores, 2019-11-08:
I guess best way would be to integrate with https://github.com/roboticslab-uc3m/vision/tree/develop/programs/tensorflowDetection2D

rsantos88, 2019-11-08:
> I guess best way would be to integrate with https://github.com/roboticslab-uc3m/vision/tree/develop/programs/tensorflowDetection2D

Yes, that's the idea! I've already been testing with the @davidvelascogarcia code and I think it could be a very interesting way to apply it in a specific interactive demo with TEO.

PeterBowman, 2021-05-01:
Now we can use object classification via DNNs: https://github.com/roboticslab-uc3m/vision/issues/103.

PeterBowman, 2021-11-19:
This is already possible thanks to our new detector architecture at the vision repo (we have [four available detector plugins](https://github.com/roboticslab-uc3m/vision/tree/a66a6c92ce5ef246c404a290596cfdf7de07fc09/libraries/YarpPlugins) at the time of writing this) and a major rework I have just merged here at https://github.com/roboticslab-uc3m/teo-follow-me/commit/a8062e56fb56bfeb52fe96c91cf686dd6ae75657. Detectors must be instantiated through the [rgbdDetection](https://github.com/roboticslab-uc3m/vision/tree/a66a6c92ce5ef246c404a290596cfdf7de07fc09/programs/rgbdDetection) app, which publishes the centroid of the closest detected object expressed in XYZ coordinates (meters). It is assumed that the origin is located at the center of the image, positive X values pointing right, positive Y values pointing down.

closed 2021-11-19 by PeterBowman